Zendure/Zendure-HA #797: [Bug] 40W output limit when max SoC is reached

cybe, 2025-09-19:
### Home Assistant Version

2025.8.2

### Zendure Integration Version

v1.1.4-pre7

### Describe the bug

Once the set maximum SoC is reached (at least via solar charging), the smart matching manager activates bypass mode but limits the output to 40 Watts. Even though the house consumption is higher, hence there is power drawn from the grid.

[image]

### To Reproduce

1. Current SoC is like 50%
2. Enable smart matching
3. Set max SoC to 90%
4. Let SoC reach 90%
5. Output limit goes to 40 Watts

### Expected Behaviour

Output limit should match p1 sensor.

### What device are you using?

SF800 Pro

### Log Extracts

```shell
Will provide, once I can reproduce it again. Battery is currently not charged enough.
```

cybe, 2025-09-21:
Thank you for looking into this, @Zanooon97. Hadn't had the chance to test things out further, yet.

Do I understand correctly, that the culprit is: https://github.com/Zendure/Zendure-HA/blob/0c0118556834a50b99e3567b14cf2c653615c5c5/custom_components/zendure_ha/device.py#L444-L445

As soon as `SOCFUL` is reached, `homeOutput` is set to `0`. Same for `solarOutput`. But it shouldn't be. And you proposed a fix for this.

Let's see how the current master is performing. The whole calculation has [changed](https://github.com/Zendure/Zendure-HA/blob/42a46c0b5abe8370491b1e09a55429eceefc6976/custom_components/zendure_ha/device.py#L424-L443) in 1e3bbbcd0fd3d3e2f7082a759908784351e90f92.

CORAAL, 2025-11-04:
Yes, the problem is still relevant (master branch 0bcea53c2e41cf4ff9b1668ac1aec27db4451771)