Zendure/Zendure-HA #803: Zendure 3CT Smart Meter Integration

MrFuellort, 2025-09-21:
### Describe the feature

Guten Tag,

ich habe eine Frage ist eine Integration des Zendure 3CT Smart Meter in die Zendure HA Integration  in naher Zukunft geplant? 
Aktuell kann ich diesen nur über MQTT in HA einbinden, aber nach jedem Neustart erhält MQTT keine Werte mehr. Erst nach mehreren Neustarts des Mosquitto Brokers kommen die Sensordaten zurück.

Was auf Dauer sehr anstrengend ist und zu fehlerhafter Datenauswertung kommt. 
Da ich diese Probleme mit der Zendure Integration nicht habe und es übersichtlicher wäre, alle Geräte an einem Ort zu haben, wäre es von Vorteil, wenn dieses Feature hinzugefügt würde.

Falls ich Ihnen mit Logs der MQTT-übermittelten Sensoren helfen kann, stelle ich diese gerne bereit.

Vielen Dank im Voraus.

### What device is this for?

SF800 Pro

harrymayr, 2025-09-27:
Ist geplant, aber erst, nachdem die 1.1.4 released ist. Da dann aber (für mich) auch noch Urlaubszeit ist, würde ich nicht vor Ende Oktober, Anfang November damit rechnen.

mimich38, 2025-11-05:
HI
I have a smartMeter 3CT from Zendure
I don't know how to detect this device  MQTT ?? or what else

sebaschn, 2025-11-07:
> Ist geplant, aber erst, nachdem die 1.1.4 released ist. Da dann aber (für mich) auch noch Urlaubszeit ist, würde ich nicht vor Ende Oktober, Anfang November damit rechnen.

+1 hier (kleiner push für dieses issue)

Bereits jetzt von uns ein großes DANKE, falls du tatsächlich Zeit dafür findest.
...und: Willkommen zurück aus dem wohlverdienten Urlaub!

MrFuellort, 2025-11-23:
Auch von mir nochmal einen großen Dank, das du dir die Zeit nimmst für die Integration.

DerUnwissende, 2025-12-09:
@harrymayr 
Guten Morgen, 
ich muss nochmal nachfragen... ich bin auf [https://github.com/Zendure/Zendure-HA/pull/939/commits](https://github.com/Zendure/Zendure-HA/pull/939/commits) gestoßen und habe mich gefragt, ob ich die Inhalte der 3 Threads kopieren muss und in config/custom-components/zendure-ha/ in den 3-Py Dateien ersetzen muss? Einfach per Copy & Paste?

MrFuellort, 2025-12-23:
Hallo Harrymayr, das werde ich nochmal versuchen, du hattest in deinem letzten Post von einer zip Datei gesprochen. 
Wo würde ich diese denn finden? 

Die fertigen Dateien zum kopieren gibt es nicht zufällig auf Github? Falls doch wo könnte ich diese finden? 

Vielen Dank und schöne Feiertage.

harrymayr, 2025-12-23:
Ein wenig weiter oben ;-)
[image]

harrymayr, 2026-03-11:
Es wird irgendwann eine neue offizielle Version geben mit der Unterstützung von Smart-Metern, aber ich kann nicht sagen wann. Wenn du magst, kannst du ja mein Release verwenden https://github.com/harrymayr/Zendure-HA/releases/tag/v1.2.4-Harry-pre4 da kannst du mit der Angabe der IP-Adresse den 3CT verwenden.

MrFuellort, 2026-05-24:
@DerUnwissende wie hast du aktuell den 3CT in Home Assistant integriert?