Zendure/Zendure-HA #806: [Bug] Illogical power distribution between two Hyper 2000 with different storage capacity

Herr-Frodo, 2025-09-22:
### Home Assistant Version

2025.5.1

### Zendure Integration Version

v1.1.4pre7

### Describe the bug

Hyper 1:  2kW PV installed, 2x AB2000X, SoC 60%
Hyper 2: 2kW PV installed, 1x AB2000X, SoC 96%

The Manager uses Hyper 1 instead of Hyper two to feed the home grid instead of Hyper 2.
Also Hyper 2 regulated down the PV input instead of feedin the grid.

### To Reproduce

.

### Expected Behaviour

The grid should be feeded from Hyper 2 in this case and Hyper 1 shoul use additional AC power (from Hyper 1) for charging.

### What device are you using?

Hyper2000

### Log Extracts

```shell

```

### Other Information

[image]

Herr-Frodo, 2025-09-22:
[image]

A few minutes later.

Zanooon97, 2025-09-27:
Probier mal diese powerdistribution, bei meinen SF funktioniert sie perfekt auch die auto balancing on full soc
https://github.com/Zendure/Zendure-HA/pull/817
https://github.com/Zanooon97/Zendure-HA/tree/master/custom_components/zendure_ha

Zanooon97, 2025-10-19:
@Apollo2022 du musst schon die dateien von meinem repo downloaden, auch switch.py

harrymayr, 2025-11-19:
@Herr-Frodo did you check out 1.1.4-pre10. Is this still an issue?

closed 2025-12-25 by github-actions[bot]