Zendure/Zendure-HA #814: [Bug] Having two AIO2400 The information are mixed or missing

Inmapi, 2025-09-24:
### Home Assistant Version

2025.5.1

### Zendure Integration Version

v1.1.3

### Describe the bug

I Have two AIO2400 , in the main screen I have the serial number of the first only but both Bluetooth addresses:

[image]

moreover:
-  the two Battery (BMS) are correctly shown separated
- but the two main controller are in some way "overlapped"

[image].

NOTE:
I will have the old AIO2400 for short time because I have to send back it. (defective MPPT  controller)

### To Reproduce

1. Get two AIO2400
2. register both on the Zendure App
3. Open the Zendure integration in HomeAssistant

### Expected Behaviour

The two AIO 2400 Units should be shown separated with their serial numbers

### What device are you using?

Aio2400

### Log Extracts

```shell

```

harrymayr, 2025-09-28:
The only idea I have is, that the both AIO have the same name in the App. But the batterie names indicate, that the new ones name is: AIO2400-New.
Can you please enable debug-logging, restart HA, wait for ~5 minutes and disable debug-logging. Perhaps I can find out with this log-file, why this happens.

Inmapi, 2025-09-30:
Hello
I followed your instructions. Find attached the debug log zipped.
BR
Max

Inmapi, 2025-10-01:
> Hi Max,
>
> there is no attached file. You can send it using ***@***.***

Hello Harrymar
hope this time the attacment arrives.
I forgot to tell that the new battery was added automatically having the
serial number as name. I renamed it for convenience.
So I expect the new controller will appear, as a new instance of AIO2400
with a conventional name, possibly the serialnumber.
BR
Max

harrymayr, 2025-10-01:
Hi Max,

from your log, I see no difference in the names of the AIO 2400. The communication with the device uses the devicekey, but the creation of the sensors in HA use the device name. So it creates a sensor i.e. sensor.aio_2400_output_home_power. And here I see the problem, if you use the same names for the devices. I propose to remove the integration in HA, rename the second AIO 2400 to AIO 2400 New, or AIO 2400 B or any other different name in the App and install the integration again.

Inmapi, 2025-10-02:
Hallo
done as you said, everithing ok,  ticket solved!
Could be useful in the readme/instructions to highlight the need to have
unique names for the devices to avoid you be annoyed as I did!.
Best regards
Max